Sergey1560/Marlin_FB4S #8: Ошибки компиляции под робин нано в.2

Malderin, 2020-09-28:
Сделал минимум изменений, но скомпилировать не могу. 
В конф.н поменял тип платы на робин нано в.2, включил вайфай в файле mks_wifi_settings.h

Исправил что смог, но вот с этим не знаю что делать:
Linking .pio\build\mks_robin_nano35\firmware.elf
.pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_ui.cpp.o:(.bss.W25QXX+0x0): multiple definition of `W25QXX'
.pio/build/mks_robin_nano35/src/src/HAL/STM32F1/eeprom_spi_w25q.cpp.o:D:\GitHub\Marlin_FB4S/Marlin\src\HAL\STM32F1/eeprom_spi_w25q.cpp:12: first defined here
c:/users/alex/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio\build\mks_robin_nano35\firmware.elf section `.bss' will not fit in region `ram'
c:/users/alex/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: region `ram' overflowed by 13824 bytes
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\mks_robin_nano35\firmware.elf] Error 1

X-Dron, 2020-09-29:
Это не ошибка компиляции, а ошибка линка.

Malderin, 2020-09-29:
> Это не ошибка компиляции, а ошибка линка. Не хватает оперативной памяти. Форк для Nano V2. лучше спрашивать у меня. https://github.com/X-Dron/Marlin_FB4S
> Сейчас там пока не актуальная версия, жду когда Серега примет PR, висящий уже 3 дня. :)
> Но даже в актуальной есть проблемы с выходом из режима передачи файлов по WiFi. Плата вешается. В работе, пробуем решить, но свободного времени очень мало.

Спасибо, будет ждать. На офф мкс что то надежды совсем мало.

X-Dron, 2020-09-29:
> Спасибо, будет ждать. На офф мкс что то надежды совсем мало.
А что не так с офф MKS? Прошивка открытая, уже заточена под Nano V2, собирается. Есть готовые бинарники с asset-ми. Другое дело, что там все заточено под их интерфейс и Wifi работает только под ним и от самого интерфейса иногда тошнит.

Malderin, 2020-09-29:
Пробовал собирать их прошивку с вайфай. Все собирается вроде, в меню появляются пункты вайфай. Можно указать название сети и пароль, только этим все и заканчивается. Заходишь в сам пункт вайфай, там вместо статуса успешного подключения и айпишника висит пункт что то вроде "подождите выполняется настройка модуля". Понятно что ничего при этом не работает.
--
Отправлено из Mail.ru для Android вторник,

Sergey1560, 2020-09-29:
> ram' overflowed by 13824 bytes

У вас не хватает памяти для сборки. Насколько я понял, вы пытаетесь собрать с интерфейсом от MKS. В МК не хватает памяти для для этого. Если вам нужен этот интерфейс, возьмите марлин либо из офф. репозитория либо на гитхабе МКС. В первом случае - вы можете использовать как образец конфиг из моей сборки, но у вас совсем не будет поддержки wifi. Кроме того, нужно переключить eeprom на sd карту (ошибка multiple definition of W25QXX).  И возможно что-то из функционала, чтобы влезло в память. После сборки свободной ram памяти должно остаться не меньше 4кб, а лучше 6.

Во втором случае (вариант с гитхаба МКС) поддержка wifi есть, но условная. Передача файлов не работает пока (https://github.com/makerbase-mks/MKS-Robin-Nano-V2/wiki/MKS_Robin_WIFI#test-wi-fi).

Malderin, 2020-09-29:
> У вас не хватает памяти для сборки. Насколько я понял, вы пытаетесь собрать с интерфейсом от MKS. В МК не хватает памяти для для этого. Если вам нужен этот интерфейс, возьмите марлин либо из офф. [...]

Да, я собираю графический вариант от мкс 
Пробовал оба варианта, и от мкс и офф марлин (как последний стабильный, так и багфикс). Результат я описывал постом выше. Возможно я что то не так делаю, но вайфай на этой плате не заводится ни в каком виде.

Sergey1560, 2020-09-29:
> Да, я собираю графический вариант от мкс
> Пробовал оба варианта, и от мкс и офф марлин (как последний стабильный, так и багфикс). Результат я описывал постом выше. Возможно я что то не так делаю, но вайфай на этой плате не заводится ни в каком виде.

Давайте я попробую пояснить еще раз: с интерфейсом от МКС wifi пока не работает. Ни в какой версии.
Я их интерфейс не использую и никаких доработок в эту сторону не делаю. Проблема связана с тем, что их интерфейс использует гигантский видео буфер для отрисовки. Его можно было бы уменьшить, но задан он в коде библиотеки LVGL которая автоматически скачивается с гитхаба МКС. Если очень нужно - можно попытаться собрать, чтобы платформио скачало библиотеку, а потом в libdeps изменить размер буфера. Но wifi все равно работать не будет. При использовании интерфейса от МКС нет никакого смысла использовать мою сборку. 

Если вам нужно запустить robin nano 2.0 с wifi, используйте вариант "New UI" из марлина, это опция TFT_480x320_SPI.

Malderin, 2020-09-29:
> Давайте я попробую пояснить еще раз: с интерфейсом от МКС wifi пока не работает. Ни в какой версии.
> Я их интерфейс не использую и никаких доработок в эту сторону не делаю. Проблема связана с тем, [...]

Теперь понял, спасибо за пояснение.

X-Dron, 2020-09-29:
Malderin, поверьте, что интерфейс от MKS не то что вам нужно. Новый "цветной" GUI-интерфейс построен на асинхронной работе с экраном, более эффективно использует ресурсы контроллера, меньше проблем с печатью из-за "нехватки быстродействия". Я обновил в своем репе до актуальной для себя. Можете попробовать собрать. Подключение Wifi работает, но фалы пока передавать не стоит. Вечером попробую поиграться с видеобуфером.

Malderin, 2020-09-29:
> Malderin, поверьте, что интерфейс от MKS не то что вам нужно. Новый "цветной" GUI-интерфейс построен на асинхронной работе с экраном, более эффективно использует ресурсы контроллера, меньше проблем с печатью из-за "нехватки быстродействия". Я обновил в своем репе до актуальной для себя. Можете попробовать собрать. Подключение Wifi работает, но файлы пока передавать не стоит. Вечером попробую поиграться с видеобуфером чтобы не было падений после передачи файла.
>   

Меня от интерфейса мкс не тошнит, как по мне он вполне себе ничего. Глубоко я конечно не копал, но в принципе для моего дры стола нехватки быстродействия я не ощутил, работает вполне себе нормально. Плюс мне нравится предпросмотр модели при печати.
Марлиновский интерфейс я пробовал ещё на старой плате робин мини, да он вполне неплох.
Попробую собрать ваш вариант с вайфай, спасибо!

X-Dron, 2020-09-29:
> Марлиновский интерфейс я пробовал ещё на старой плате робин мини, да он вполне неплох.
> Попробую собрать ваш вариант с вайфай, спасибо!

Забыл предупредить, :) там включен BLTouch и 3 драйвера по UART. Думаю разберетесь.

Malderin, 2020-09-29:
У меня как раз блтач и все 2209 по юарт
--
Отправлено из Mail.ru для Android вторник,

X-Dron, 2020-09-29:
> У меня как раз блтач и все 2209 по юарт

Успел с скачать в 2.0.6.1? А то мы уже обновились на 2.0.7. Я конфиг под nano еще не заливал. Вечером проведу тест, обновлю.

Malderin, 2020-09-29:
Не успел, я на работе, все манипуляции будут только вечером
--
Отправлено из Mail.ru для Android вторник,

X-Dron, 2020-09-29:
> Не успел, я на работе, все манипуляции будут только вечером 

Если что, то старый сорс здесь

Malderin, 2020-09-29:
> > Не успел, я на работе, все манипуляции будут только вечером
> 
> Если что, то старый сорс, покопаться, здесь
> https://yadi.sk/d/Mjsgndkfd5N1ow

Попробовал. не менял вообще ничего, кроме файла mks_wifi_settings.h (вбил название и пароль к своей сети и раскомментировал #define MKS_WIFI_ENABLED_WIFI_CONFIG).
При включении в нижней строке пишет WIFI:waiting...
Понятно, что не соединяется, непонятно почему. Подожду версию 2.0.7 из вашей ветки и попробую собрать поновой.

X-Dron, 2020-09-29:
> Попробовал. не менял вообще ничего, кроме файла mks_wifi_settings.h (вбил название и пароль к своей сети и раскомментировал #define MKS_WIFI_ENABLED_WIFI_CONFIG).
> При включении в нижней строке пишет [...]

Странно как-то это все. Wifi-модуль вообще установлен? Фотку можно? В Wifi-модуль прошивка от MKS залита? Может в этом проблема, а не в Marlin-е. Такое ощущение, что плате не видит модуля. Если между ними связь есть, но Wifi не может никуда подключиться, то он включается в режим точки доступа и сообщает об этом. Точки доступа тоже не появляется?

Malderin, 2020-09-29:
Модуль, разумеется был установлен. Но ваша правда, передёрнул его и все заработало, видимо где то был плохой контакт.
--
Отправлено из Mail.ru для Android вторник,

Malderin, 2020-09-29:
> > Попробовал. не менял вообще ничего, кроме файла mks_wifi_settings.h (вбил название и пароль к своей сети и раскомментировал #define MKS_WIFI_ENABLED_WIFI_CONFIG).
> > При включении в нижней строке пишет WIFI:waiting...
> > Понятно, что не соединяется, непонятно почему. Подожду версию 2.0.7 из вашей ветки и попробую собрать поновой.
> 
> Странно как-то это все. Wifi-модуль вообще установлен? Фотку можно? В Wifi-модуль прошивка от MKS залита? Может в этом проблема, а не в Marlin-е. Такое ощущение, что плате не видит модуля. Если между ними связь есть, но Wifi не может никуда подключиться, то он включается в режим точки доступа и сообщает об этом. Точки доступа тоже не появляется?
> В своем репе обновил конфиг до актуальной версии на 2.0.7. Все собирается, ошибка с зависоном платы после передачи файла исправлена. Появилась другая. Нет автостарта печати после передачи. Серега что-то недокопировал при переходе на обновление. Но это уже не критичная ошибка.

Я понял все что вы думаете по поводу графической прошивки от мкс, но все таки, реально сделать вайфай с их интерфейсом?

X-Dron, 2020-09-30:
> Я понял все что вы думаете по поводу графической прошивки от мкс, но все таки, реально сделать вайфай с их интерфейсом?
> Пробовал собирать новую версию марлина 2.0.7 - выдает ту же ошибку с нехваткой памяти:
> Linking .pioenvs\mks_robin_nano35\firmware.elf
> .pioenvs/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/irq_overrid.cpp.o: In function `__irq_usart1': D:\GitHub\Marlin_2.0.x_official/Marlin\src\lcd\extui\lib\mks_ui/irq_overrid.cpp:47: multiple definition of `__irq_usart1'
> .pioenvs/mks_robin_nano35/src/src/HAL/STM32F1/MarlinSerial.cpp.o:D:\GitHub\Marlin_2.0.x_official/Marlin\src\HAL\STM32F1/MarlinSerial.cpp:114: first defined here
> collect2.exe: error: ld returned 1 exit status
> *** [.pioenvs\mks_robin_nano35\firmware.elf] Error 1

1. Это не ошибка нехватки памяти, а ошибка многократного определения функции __irq_usart1. MKS, похоже, опять натюхарили в официальный марлин какую-то непотребщину. На 2.0.6.1 такого не было
2. Если хочешь Wifi и MKS-интерфейс, то единственный путь [сюда](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware)
3. Раздели мух и котлет. Добейся работы Wifi модуля на прошивке, где он 100% работает. У тебя Wifi-модуль прошит бинарником от MKS (MksWifi.bin)?

Malderin, 2020-09-30:
> 2. Если хочешь Wifi и MKS-интерфейс, то единственный путь [сюда](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware)
Пробовал вчера, собирается, осями можно подвигать и все такое. Файлы, понятное дело, не передаются - выдается ошибка.

> 3. Раздели мух и котлет. Добейся работы Wifi модуля на прошивке, где он 100% работает. У тебя Wifi-модуль прошит бинарником от MKS (MksWifi.bin)?
Нет не прошивал, они пишут в своем вики что это не требуется:
https://github.com/makerbase-mks/MKS-Robin-Nano-V2/wiki/MKS_Robin_WIFI
"Tips: MKS Robin WIFI has uploaded firmware on the factory, so you need't update the firmware again"

Malderin, 2020-09-30:
Честно сказать пробовал подкидывать бинарник на карту памяти - ничего не происходит при включении.

closed 2020-10-05 by Sergey1560