Sergey1560/Marlin_FB4S #9: Баг с осью Z

Malderin, 2020-10-04:
Принтер на плате робин нано в.2 (не медведь). Стоит блтач от треугольников.
Запустил на печать обычный джикод (перед каждой печатью стоит G29).
Перед каждой печатью стал замечать странное поведение оси z. В одном месте образовался прямоугольник, в котором ось z с каждым новым слоем опускалась вниз (ниже будут фото). В остальной области печати все шло нормально.
Из последних изменений, что я сделал, это включил junction deviation вместо классических рывков (всегда их использовал на всех прошивках без проблем) и отключил двойную пробу при построении карты стола.
Что такое может быть?

Фото:
https://yadi.sk/i/vnaqordL7SbVjg
https://yadi.sk/i/8iQiQ0jUIIV2Pw
https://yadi.sk/i/ozcjXiEytpasLw

Остановил печать когда голова была как раз в этой зоне. Оси x y двигаются, ось z на команды из меню принтера не реагирует никак.
Отправил z в дом - после этого она заработала.

Malderin, 2020-10-04:
на всякий случай джикод, который я печатать пытался, этот дефект было видно сразу уже при печати юбки
https://yadi.sk/d/hNKfI-5wRrPPaw
Для проверки, прошил принтер еще раз, запустил его же, все повторилось.

Malderin, 2020-10-04:
И заодно, чтобы не плодить запросы, спрошу здесь.
Прописал в платформио ини айпишник своего принтера (тот что светится на экране когда соединение установлено). Прописал curl в переменные среды винды в разделе path. При вводе команд curl через консоль винды получаю нормальные ответы, т.е. curl в системе видно нормально. 
А вот прошивка не хочет загружаться в принтер, компилируется но не закидывается на карту, и следовательно принтер не перезагружается.
Эта фича просто не работает или я что то не так делаю?

Sergey1560, 2020-10-05:
> И заодно, чтобы не плодить запросы, спрошу здесь.
> Прописал в платформио ини айпишник своего принтера (тот что светится на экране когда соединение установлено). Прописал curl в переменные среды винды в разделе path. При вводе команд curl через консоль винды получаю нормальные ответы, т.е. curl в системе видно нормально.
> А вот прошивка не хочет загружаться в принтер, компилируется но не закидывается на карту, и следовательно принтер не перезагружается.
> Эта фича просто не работает или я что то не так делаю?
> 
> p.s. печать через вайфай модуль идет нормально, файлы из куры закидываются и печать стартует.

Исправил скрипт платформио, отправка должна работать.

Sergey1560, 2020-10-05:
> на всякий случай джикод, который я печатать пытался, этот дефект было видно сразу уже при печати юбки
> https://yadi.sk/d/hNKfI-5wRrPPaw
> Для проверки, прошил принтер еще раз, запустил его же, все повторилось.

Попробуйте сделать авто-калибровку командой G29, а потом посмотреть командой [M420](https://marlinfw.org/docs/gcode/M420.html) результаты. Возможно в этом месте стола замеры значительно отличаются от остальных.

Malderin, 2020-10-05:
> Исправил скрипт платформио, отправка должна работать.

Да, теперь отлично, спасибо.

closed 2020-10-07 by Malderin