Sergey1560/Marlin_FB4S #10: Проблема WIFI?

Malderin, 2020-10-08:
Плата мкс робин нано в.2.
Предыстория. Вчера-сегодня принтер запорол 2 печати. Оба раза я был возле него и видел что именно произошло. Первый раз , когда модель допечаталась примерно до 40% голова по z поехала вниз прямо в модель. Естественно модель отлетела от стола и примерно 1 ч 15 мин печати было запорото. 
Что я сделал. Переслайсил модель по новой, зкинул на карту памяти. Аналогичная проблема мне была знакома по моей старой плате (мкс робин мини), правда там в ноль ехали оси х и у, но это ничего не меняет, так тогда я подумал и сделал вот что.
Убавил частоту SDIO c дефолтной 4.5 МГц до 3, увеличил число опросов SDIO с 3 до 5 (все это сделал с файле пинс):
#define SDIO_CLOCK                       3000000
#define SDIO_READ_RETRIES                  5
Сегодня, залил эту вновь собранную прошивку , закинул вновь отслайсенный файл запустил печать. И опять то же самое. При чем в полном смысле, сопло было в том же месте и на той же высоте. Поразительное совпадение, далее расскажу свои предположения по этому поводу. 
Недавно заметил вот что. Была запущена кура (она соотвественно настроена на вайфай соединение). Потом, спустя какое то время не закрывая куру, открыл атом поделал кой че, собрал прошивку с опцией заливки сразу в принтер. Сразу заметил что прошивка льется с гораздо меньшей скоростью и не заливается до конца, где то как раз процентах на 40 все останавливается и спустя немного времени принтер сообщает об ошибке. Вспомнил что кура запущена, закрыл и сразу все нормализовалось. 
Предположение №1: вчера, когда я печатал первый раз, кура была запущена. Я подумал что возможно из за соединения с курой произошло падение скорости или полный обрыв связи с картой памяти, что и вызвало такое поведение, принтер поехал в ноль.
Сегодня когда печатал второй раз, все было отключено т.е. ни кура ни какие то другие программы, держащие соединение с принтером запущены не были. Все повторилось. 
Предположение №2: что то не так с поднятым соединением вайфай просто с роутером, что вызывает аналогичное поведение (падение скорости или полный обрыв связи с картой памяти, что и вызвало такое поведение, принтер поехал в ноль).
Несколько фоток:
[Запоротые детали](https://yadi.sk/i/51ZRbhc-IT73Mw)
[Еще фото](https://yadi.sk/d/4XpDbinoe5O7Ag)
[Тот самы джикод, котрый я печатал второй раз](https://yadi.sk/d/f3QgYCh-AONA0g)

К слову джикод просмотрел и ничего криминального не нашел, вполне нормальный код, никаких пропусков или Z в минус я там не обнаружил. Высота сбоя 23-25 мм, на случай если захотите сами проверить мои слова.
Третья попытка на собранном последнем офф марлине (багфикс от 08.10.2020) - все отпечатолось сразу и без проблем.

Sergey1560, 2020-10-09:
1. Прошивка  MKS WIFI модуля сделана так, что возможно только одно соединение. Если открыта Cura, загрузка прошивки из platformio работать не будет. Это ограничение прошивки именно esp модуля.

2. Во время печати все данные, которые приходят от WIFI модуля, игнорируются и в обработчик марлина не попадают. Пока идет печать, wifi модуль никак влиять не должен.

3. Уменьшение скорости SDIO или увеличение количества попыток это костыль, который в данный момент не нужен. Когда-то давно, в драйвере SDIO марлина был баг, который приводил к проблемам чтения с карты, но его исправили. Если есть ошибки чтения с карты, они в каком-то другом месте и исправлять их нужно там, а не уменьшать скорость чтения и увеличивать количество попыток.

4. G-код на первый взгляд адекватный, каких-то проблем я не заметил.

Я могу предположить только какие-то проблемы с чтением с SD карты. В какой-то момент читается мусор. Попробуйте отформатировать карту или с другой картой.

Malderin, 2020-10-10:
Спасибо!

closed 2020-10-10 by Malderin