Sergey1560/Marlin_FB4S #13: All drivers A4988?

chrisdaloa, 2020-10-28:
Hi, I would to compile my own firmware starting from your one.
In Configuration.h I see that all drivers are configured as A4988. 
I have a Ghost 5 stock so I suppose I have to change X and Y driver type in TMC2208. Is it right or should I use TMC2208_STANDALONE type?
And should I change something also in the "tmc smart" section in configuration_adv.h ?
Thanks a lot!

Chazy12, 2020-10-28:
For the stock ghost 5, you just have to undefine ​//​#define FB_5_STOCK and define ​#​define​ ​ALL_DRV_2208​.

So should be like this:

//​#​define​ ​ALL_DRV_2208​
​//​#define FB_4S_STOCK​
​​#define FB_5_STOCK

Sergey1560, 2020-10-29:
> Hi, I would to compile my own firmware starting from your one.
> In Configuration.h I see that all drivers are configured as A4988.
> I have a Ghost 5 stock so I suppose I have to change X and Y driver [...]

"tmc smart" section is for drivers connected to uart. If you didn't add this, you don't need these parameters. It is enough to set FB_5_STOCK, all other parameters related to drivers will be set. 

If the driver is not connected via uart (stock ghost5), it makes no difference which mode to specify - TMC2208_STANDALONE or A4988.

closed 2020-10-29 by Sergey1560