Sergey1560/Marlin_FB4S #20: Русский язык не поддерживается?

Engi23, 2020-11-28:
Вместо русского языка кракозяблики

Sergey1560, 2020-11-28:
На данный момент для варианта интерфейса TFT_COLOR_UI не поддерживается. В шрифте нет кириллических символов. 
В интерфейсе TFT_CLASSIC_UI поддержка русского языка есть.

Engi23, 2020-11-30:
Да меня интересует TFT_COLOR_UI Это же точечный дисплей, значит шрифт должен быть набором массивов точек. Может добавить? Шрифт же в отдельном файле лежит? Если дадите пример как зашифровать 1 букву, думаю я смогу сделать для остальных.

Sergey1560, 2020-11-30:
Пример шрифта - Marlin/src/lcd/tft/fontdata/helvetica_18.cpp
В комментариях там описано, в каком виде лежит массив.
Есть еще одно небольшое ограничение - марлин использует максимум 255 символов.

Engi23, 2020-12-01:
Вот этот файл. https://github.com/Sergey1560/Marlin_FB4S/blob/FB4S_WIFI/Marlin/src/lcd/tft/fontdata/helvetica_18.cpp к сожалению не могу понять как символы закодированны в массиве. Похоже разные буквы занимают разный объем, какой там символ разделитель и как закодированы пиксели?

Sergey1560, 2020-12-06:
Описание структуры font_t находится в Marlin/src/lcd/tft/tft_string.h

closed 2020-12-21 by Sergey1560

Engi23, 2020-12-22:
TFT_CLASSIC_UI поддерживает сенсорный ввод?

Sergey1560, 2020-12-22:
> TFT_CLASSIC_UI поддерживает сенсорный ввод?

Да, поддерживает.