Sergey1560/Marlin_FB4S #21: Probe wizard не работает

Engi23, 2020-11-30:
Запускаю Probe wizard происходит homing xyz после чего пустой синий экран и принтер выключается от mks pwc модуля.
probe настроен на вход Z max. 
Z home на вход Z min

Sergey1560, 2020-11-30:
Если вы обратите внимание, то при сборке с включенным PROBE_OFFSET_WIZARD и TFT_COLOR_UI компилятор сообщает вам:
```
.pio/build/mks_robin_nano35/src/src/lcd/menu/menu_probe_offset.cpp.o: In function `probe_offset_wizard_menu()':
/home/sergey/Projects/3D_Printer/FirmWare/Marlin/Marlin2.0/Marlin/src/lcd/menu/menu_probe_offset.cpp:93: warning: undefined reference to `lcd_moveto(unsigned char, unsigned char)'
.pio/build/mks_robin_nano35/src/src/lcd/menu/menu_probe_offset.cpp.o: In function `lcd_put_u8str(char const*)':
/home/sergey/Projects/3D_Printer/FirmWare/Marlin/Marlin2.0/Marlin/src/lcd/menu/../lcdprint.h:161: warning: undefined reference to `lcd_put_u8str_max(char const*, unsigned short)'
```
Это означает, что функции lcd_moveto и lcd_put_u8str_max вызваются, но они не существуют. 
В результате происходит вызов функции по адресу 0, МК попадает в hard fault и остается там навсегда. Срабатывает watchdog и МК перезагружается. Бутлоадер не удерживает ногу управления MKS PWC, поэтому плата просто выключается.

PROBE_OFFSET_WIZARD - новая функция и она еще не реализована в TFT_COLOR_UI.

Mikhail-Kotov, 2020-11-30:
Решается заменой файла menu_probe_offset.cpp на вот этот
https://raw.githubusercontent.com/rhapsodyv/Marlin/7d1e66abdb2e661b916d2af50c2d844988a338da/Marlin/src/lcd/menu/menu_probe_offset.cpp

Sergey1560, 2020-11-30:
> Решается заменой файла menu_probe_offset.cpp на вот этот
> https://raw.githubusercontent.com/rhapsodyv/Marlin/7d1e66abdb2e661b916d2af50c2d844988a338da/Marlin/src/lcd/menu/menu_probe_offset.cpp

Насколько я понимаю, это от одного из основных разработчиков марлина и конкретно color_ui. Этого в bugfix ещё нет?

Mikhail-Kotov, 2020-11-30:
В bugfix вроде бы уже поновее даже сделали
https://github.com/MarlinFirmware/Marlin/pull/20239

У меня после замены файла Wizard заработал. Брал я отсюда, тестировал дней 10 назад.
https://github.com/Speaka/Marlin/blob/90624eb3f348b5f82a7693e918314b08c5e4e4f3/Marlin/src/lcd/menu/menu_probe_offset.cpp

Sergey1560, 2020-12-06:
В данном релизе этот функционал для COLOR_UI еще не реализован.
Вероятно в 2.0.8 будет добавлено.

closed 2020-12-06 by Sergey1560