Sergey1560/Marlin_FB4S #22: При загрузке по wifi плата часто выключается

Engi23, 2020-12-06:
При загрузке по wifi плата часто выключается. Особенно если включен преднагрев. Недавно когда экструдер остывал - температура 160-60 установлено 0, принтер выключался раз 5 подряд. Разумеется установлена плата mkspwc с помощью которой плата и выключается

Sergey1560, 2020-12-06:
Опишите как воспроизвести проблему. 
Файл большой?
Отправляете из Cura? 
Проявляется только при включенном преднагреве?
Собирали сами, или используете собранную прошивку?

Engi23, 2020-12-08:
Проявляется при разных файлах, иногда даже маленьких. Если включить преднагрев особенно для abs 100 / 245 и попытаться отправить большой файл проблема появляется чаще. Прошивку собирал сам. Отправляю через cura. Нет иногда выключается и без преднагрева, но сильно реже. Может происходит ошибка при передачи по wifi и от этого он выключается? Принтер стоит за стеной в соседней от роутера комнате. Хотя раньше хоть ошибки передачи случались но не настолько часто насколько сейчас выключения. Возможно кажется из за того что выключение неприятнее просто ошибки передачи.

Engi23, 2020-12-17:
Гарантированный способ повторения проблемы начать передачу по wifi и обесточить роутер.

Sergey1560, 2020-12-18:
> Гарантированный способ повторения проблемы начать передачу по wifi и обесточить роутер.

Проблема в том, что watchdog сбрасывается в момент приема данных от esp. Период сброса WD примерно 4 секунды. Если за это время сброса не произошло, WD перезагрузит плату. 
Обычно, от esp пакеты идут гораздо быстрее и проблем нет. В вашем случае, видимо по какой-то причине esp прекращает передачу больше чем на 4 секунды. 
Я думаю, это можно исправить, постараюсь сделать.

Sergey1560, 2020-12-21:
А у вас актуальная версия? В файле Marlin/src/module/mks_wifi/mks_wifi_sd.cpp строка 202,  iwdg_feed(); сбрасывает watchdog достаточно часто, даже если с передачей проблемы.

Engi23, 2020-12-22:
файл датирован 18,10,2020, но строчка другая. Тут написано что последнее обновление это файла 2 месяца назад. В общем скачал прошил посмотрим как будет работать. Ваш файл после скачивания датируется 30,11,2020...

Sergey1560, 2020-12-22:
> файл датирован 18,10,2020, но строчка другая. Тут написано что последнее обновление это файла 2 месяца назад. В общем скачал прошил посмотрим как будет работать. Ваш файл после скачивания датируется 30,11,2020...

Пожалуйста, обновите весь код. Рекомендую разобраться, как работает git и как обновлять проект.

Engi23, 2020-12-28:
Где почитать про то как обновлять? Я просто все скачал и вставил свои конфиги. Ну и опять пришлось MKSPWC включать в недрах прошивки. Кстати не знаю к кому эта "претензия"
но Marlin\src\pins\stm32f1\pins_MKS_ROBIN_NANO_V2.h
зачем засовывать #ifdef MKSPWC внутрь  #if HAS_TFT_LVGL_UI? 
#if HAS_TFT_LVGL_UI
  //#define MKSPWC
  #ifdef MKSPWC
    #define SUICIDE_PIN                     PB2   // Enable MKSPWC SUICIDE PIN
    #define SUICIDE_PIN_INVERTING          false  // Enable MKSPWC PIN STATE
    #define KILL_PIN                        PA2   // Enable MKSPWC DET PIN
    #define KILL_PIN_STATE                  true  // Enable MKSPWC PIN STATE
  #endif

То есть я на вашей прошивке просто так включить MKSPWC раскомментировав  //#define MKSPWC не могу.

Sergey1560, 2020-12-28:
Вам нужно клонировать репозиторий, создать свою ветку, внести туда нужные вам изменения. Для обновления нужно сделать pull основной ветки и потом merge в вашу. Так вы будете получать обновления, сохраняя ваши изменения.
В противном случае, вам каждый раз придется вносить ваши изменения вручную.

Это все не имеет никакого отношения с теме данного issue, поэтому я его закрываю.

closed 2020-12-28 by Sergey1560