Sergey1560/Marlin_FB4S #23: Не прошивает модуль wifi

Kost38, 2020-12-07:
Прошивка из релиза не прошивает модуль wi-fi. Нет никакой реакции при загрузке на файл MksWifi.bin на карточке. Файл пробовал разный, 300 кб и 4 мб.

Sergey1560, 2020-12-07:
В прошивке на данный момент нет загрузчика для esp модуля. На практике, он не очень нужен. Стандартный модуль уже прошит и необходимости менять в нем прошивку не возникает. В случае установки самодельного модуля, его проще прошить "на столе", до установки в плату. Эта функция была не очень нужна и я ее не реализовывал.

Судя по коммитам в bugfix ветку, МКС добавляет к своему интерфейсу поддержку WIFI модуля. В том числе и обновление прошивки в нем. Таким образом, в скором времени можно ожидать поддержку этого в стабильной ветке Marlin.

closed 2020-12-07 by Sergey1560

Kost38, 2020-12-07:
Спасибо за объяснение, теперь все понятно.