Sergey1560/Marlin_FB4S #29: Не работает выгрузка филамента

Ustinych, 2021-01-28:
Не работает выгрузка филамента

Sergey1560, 2021-01-28:
Будьте добры, немного подробностей.

Ustinych, 2021-01-28:
Когда в меню выбираешь выгрузить филамент, принтер вытягивает пруток примерно на 30 мм и сразу втягивает его обратно. В релизных прошивках такого  не было.

Engi23, 2021-02-02:
а не забыли ли вы правильно настроить датчик филамента? например инвертировать его состояние. Очень похоже на что то такое

Ustinych, 2021-02-08:
Датчик филамента настроен корректно. В настройках изменил параметр #define FIL_RUNOUT_STATE на LOW

Sergey1560, 2021-03-21:
> Датчик филамента настроен корректно. В настройках изменил параметр #define FIL_RUNOUT_STATE на LOW

Настройка датчика филамента зависит от конкретной модели принтера. Для FB4S и FB5 это значение отличается. В конфигурации стоит значение для FB4S.

closed 2021-03-21 by Sergey1560