Sergey1560/Marlin_FB4S #31: Расширение функционала Marlin на FBG5

Griiis, 2021-02-19:
Добрый день, хотел бы занять для себя (и в перспективе может ещё кто захочет так же заморочиться), небольшим расширением Марлина в плане его разговорчивости. Если есть возможность - хотелось бы добавить отправление информации по COM порту на определенные операции (обрыв филамента, файл принят, файл не принят, печать окончена, печать начата, и т.д.), что бы на другой стороне микроконтроллер с небольшой оснасткой в виде MP3 плеера/синтезатора голоса и колонки мог, приняв эти данные, озвучить действие голосом. Если в Marlin уже есть отправка таких команд по COM - буду благодарен, если укажите на них. Если нет - то буду благодарен, если укажете, в какую часть кода нужно смотреть, что бы добавить этот функционал в прошивку.

Sergey1560, 2021-02-19:
Добрый день.

На большинство событий Marlin что-то сообщает в терминал. Подключите его по USB (последовательный интерфейс, скорость 115200) и посмотрите вывод в терминал сообщений на нужные Вам события.
Если на какое-то из нужных Вам событий сообщений нет, вы можете их добавить самостоятельно используя функцию SERIAL_ECHOLN().

Griiis, 2021-02-19:
> Добрый день.
> 
> На большинство событий Marlin что-то сообщает в терминал. Подключите его по USB (последовательный интерфейс, скорость 115200) и посмотрите вывод в терминал сообщений на нужные Вам события.
> Если на какое-то из нужных Вам событий сообщений нет, вы можете их добавить самостоятельно используя функцию SERIAL_ECHOLN().
> 
> Есть ещё другой вариант решения - скриптами постобработки добавить в слайсере в gcode команду [M118](https://marlinfw.org/docs/gcode/M118.html) с нужными Вам сообщениями, а в терминале их получать и обрабатывать как Вам нужно.

Благодарю, попробую поэкспериментировать, если совсем никак будет не получаться - отпишусь сюда.

closed 2021-02-28 by Sergey1560