Sergey1560/Marlin_FB4S #35: Графический интерфейс от MKS

Dima-911, 2021-03-15:
Графический интерфейс от MKS по Вашей ссылке https://sergey1560.github.io/fb4s_howto/mks_ui/ не компилируется

Verbose mode can be enabled via `-v, --verbose` option
Copying MKS Assets
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103VE.html
PLATFORM: ST STM32 (6.1.1) > STM32F103VE (64k RAM. 512k Flash)
HARDWARE: STM32F103VET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, jlink, stlink)   
PACKAGES:
 - framework-arduinoststm32-maple 2.10000.200103 (1.0.0)      
 - tool-stm32duino 1.0.2
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareSerialM> 1.0.0
|-- <lvgl> 6.1.1
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.91
|-- <Wire> 1.0
|-- <FreeRTOS701>
|-- <Servo(STM32F1)> 1.1.2
Building in release mode

warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=0 to SConscript calls.
Missing SConscript 'buildroot\share\PlatformIO\scripts\common-dependencies-post.py'
File "C:\Users\Dima\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 185, in <module>
Compiling .pio\build\mks_robin_nano35\FrameworkArduinoVariant\board.cpp.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduinoVariant\wirish\boards.cpp.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduinoVariant\wirish\boards_setup.cpp.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduinoVariant\wirish\start.S.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduinoVariant\wirish\start_c.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduinoVariant\wirish\syscalls.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\Print.cpp.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\WString.cpp.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\cxxabi-compat.cpp.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\ext_interrupts.cpp.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\hooks.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\itoa.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\libmaple\adc.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\libmaple\adc_f1.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\libmaple\bkp_f1.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\libmaple\dac.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\libmaple\dma.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\libmaple\dma_f1.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\libmaple\exc.S.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\libmaple\exti.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\libmaple\exti_f1.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\libmaple\flash.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\libmaple\fsmc_f1.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\libmaple\gpio.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\libmaple\gpio_f1.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\libmaple\i2c.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\libmaple\i2c_f1.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\libmaple\iwdg.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\libmaple\nvic.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\libmaple\pwr.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\libmaple\rcc.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\libmaple\rcc_f1.c.o
Compiling .pio\build\mks_robin_nano35\FrameworkArduino\libmaple\spi.c.o
[... 54 lines cut ...]
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:706:4: error: #error "LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Character LCD, Graphical LCD, TFT, or EXTENSIBLE_UI."
   #error "LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Character LCD, Graphical LCD, TFT, or EXTENSIBLE_UI."
    ^~~~~
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:812:6: error: #error "FILAMENT_RUNOUT_SENSOR requires FIL_RUNOUT_PIN."
     #error "FILAMENT_RUNOUT_SENSOR requires FIL_RUNOUT_PIN."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\SPI.cpp:43:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:706:4: error: #error "LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Character LCD, Graphical LCD, TFT, or EXTENSIBLE_UI."
   #error "LCD_SET_PROGRESS_MANUALLY requires LCD_PROGRESS_BAR, Character LCD, Graphical LCD, TFT, or EXTENSIBLE_UI."
    ^~~~~
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:812:6: error: #error "FILAMENT_RUNOUT_SENSOR requires FIL_RUNOUT_PIN."
     #error "FILAMENT_RUNOUT_SENSOR requires FIL_RUNOUT_PIN."
      ^~~~~
*** [.pio\build\mks_robin_nano35\src\src\HAL\STM32F1\HAL.cpp.o] Error 1
*** [.pio\build\mks_robin_nano35\src\src\HAL\STM32F1\HAL_SPI.cpp.o] Error 1
*** [.pio\build\mks_robin_nano35\src\src\HAL\STM32F1\MarlinSerial.cpp.o] Error 1
*** [.pio\build\mks_robin_nano35\src\src\HAL\STM32F1\SPI.cpp.o] Error 1
================================================================================================== [FAILED] Took 19.99 seconds ==================================================================================================
Environment       Status    Duration
----------------  --------  ------------
mks_robin_nano35  FAILED    00:00:19.992
============================================================================================= 1 failed, 0 succeeded in 00:00:19.992 =============================================================================================Процесс терминала "C:\Users\Dima\.platformio\penv\Scripts\platformio.exe 'run'" был завершен с кодом выхода 1.

Терминал будет повторно использоваться задачами. Чтобы закрыть его, нажмите любую клавишу.

Ilya-Pak, 2021-03-19:
Вроде в логе все написано, что не так.
LCD_SET_PROGRESS_MANUALLY не поддерживается с ROBIN_NANO_TFT35
И FILAMENT_RUNOUT_SENSOR требует установку FIL_RUNOUT_PIN.

Sergey1560, 2021-03-21:
Это описание сборки под MKS UI устарело. Есть готовая конфигурация под FB4S и FB5 - https://github.com/FBGhost-community/Marlin

closed 2021-03-21 by Sergey1560