Sergey1560/Marlin_FB4S #37: Marlin bugfix-2.0 with wifi

anton00706, 2021-04-06:
Hello Sergey.
Do you plan to fork official Marlin bugfix-2.0 and add wifi for Flying bear motherboard?
With TFT color UI.

Sergey1560, 2021-04-07:
Hello.
This is a stable Marlin branch (2.0.x) and it works with wifi (with COLOR_UI).
If you are interested in MKS UI (LVGL_UI), fork for a Flyingbear based on the bugfix branch is located here -  https://github.com/FBGhost-community/Marlin

anton00706, 2021-04-23:
I compiled the firmware from https://github.com/FBGhost-community/Marlin but Wi-Fi doesn't work as it should.
I tried to transfer files from Cura and MKS_WIFI_PS_upload.exe with different filename length. After transferring all files have .GCO extension and they are not printing. I attached my config and a screenshot with files on SD. If I copy files directly to SD they have .gcode extension and they are printing well.
[Marlin.zip](https://github.com/Sergey1560/Marlin_FB4S/files/6366877/Marlin.zip)
[image]

Sergey1560, 2021-05-07:
This fork has nothing to do with MKS UI.

closed 2021-05-07 by Sergey1560