Sergey1560/Marlin_FB4S #39: Запуск принтера

KuzmaVladimirovich, 2021-05-23:
Здравствуйте, Сергей! Я в 3д печати человек новый, во многом пока что не разбираюсь, обращаюсь к Вам за советами.
Преобрел себе принтер Flying Beaf Ghost 5 и понемногу его модернизирую, а именно поменял шумные драйверы на ТМС 2209, поставил модуль MKS PWC v3.0, все механически установил, осталось программно все доделать и на этом я застрял. Прошивку пытаюсь накатить, но нигде не нашел нужной информации, что именно скачивать, как устанавливать, только понял, что под драйверы нужно специальный файл скачать. У Вас на главной странице есть раздел Marlin_FB4S, внутри него 12 файлов и 7 папок. Скачал их все+файл прошивки под нужные драйвера, закинул все это на флешку в корне и вставил в принтер. Прошивка, "как бы" установилась, появляется заставка Марлин и принтер сразу же отключается, и так каждый раз при включении. Подскажите пожалуйста, может быть я что-то делаю не так?
Также у Вас видел статью про модуль MKS PWC v3.0, хочу сделать так же, весь электромонтаж произвел, осталось внести корректировки в прошивку. Может причина отключения принтера в этом модуле? Монтаж произвел согласно схеме

MCZarin, 2021-05-24:
Если вы скачали файл из раздела Releases - это уже собранная прошивка, и на флешку достаточно кинуть только этот файл. Но тогда вы получите прошивку, сконфигурированную Сергеем, без возможности изменения ключевых параметров принтера.
Да, вполне возможно, что принтер перезагружается из-за модуля PWC. Как вариант - попробуйте временно запустить принтер без него, чтобы решать проблемы последовательно.
Файлы и папки, которые вы видите на главной странице (раздел Code) - это не сама прошивка, а её исходники (текст программы, написанный людьми). Чтобы получить прошивку, её надо сначала сконфигурировать под свои нужды (тот же PWC, например), а затем скомпилировать (преобразовать в машинный код, который "понятен" процессору в принтере).
Для добавления функций PWC вам скорее всего понадобится идти именно этим путем.
Посмотрите инструкцию от Сергея: https://github.com/Sergey1560/Marlin_FB4S#как-работает-как-настроить
В ней упоминается видео от Дмитрия Соркина, а далее важные нюансы и отличия от этого видео. На шаге выкачки исходников прошивки качайте их не из репозитария Marlin, а отсюда, из репозитария Сергея. Основная часть параметров уже сконфигурирована для FBG 5, поэтому советы по конфигам из ролика Дмитрия повторять не нужно. Надо только обратить внимание на параметры, относящиеся к драйверам (вроде как уже выбраны 2209, но лучше перепроверить по инструкции) и к PWC (тут я сам не в теме).

Mikhail-Kotov, 2021-05-24:
У меня похожая проблема, плата MKS Robin Nano V2 на Sapphire Pro (CoreXY). После заставки сразу же на красном фоне говорит: KILLED. PRINTER HALTED. PLEASE RESET.
Правда это только наблюдаю на том Marlin FB4S 2.0.8.1, что у вас. Официальный Marlin 2.0.8.1 запускается, хоть и без поддержки SD.
При этом ваша же версия 2.0.7.2 в целом без особых проблем работала с перенастроенным под мои нужды конфигом.
На плате вставлен модуль MKS WiFi.
Может быть эта проблема связана с кодом работы с EEPROM ?

MCZarin, 2021-05-24:
> Правда это только наблюдаю на том Marlin FB4S 2.0.8.1, что у вас. Официальный Marlin 2.0.8.1 запускается, хоть и без поддержки SD.
> При этом ваша же версия 2.0.7.2 в целом без особых проблем работала с перенастроенным под мои нужды конфигом.

Между 2.0.7.2 и 2.0.8.1 в репо Сергея был такой коммит: "* Merge pull request #38 from X-Dron/PR1 Update pins_MKS_ROBIN_NANO_V2.h"
Там как раз в файле pins_MKS_ROBIN_NANO_V2.h (ваша плата) был добавлен код:
#define KILL_PIN                          PA2
#define KILL_PIN_STATE                    true
Думаю, это изменение и привело к такому поведению (прошивка думает, что ей на вход всегда поступает сигнал KILL). Если у вас в принтере нет PWC или его аналога, нужно либо убрать эти строки, либо посмотреть конфиги, возможно там оказался включен функционал, которого у вас нет в "железе" (из-за чего прошивка думает, что сработал .

Mikhail-Kotov, 2021-05-24:
Действительно именно это и привело. Закоментировал эти 2 строчки и всё заработало. Спасибо огромное за быструю подсказку!

Sergey1560, 2021-05-25:
@MCZarin спасибо за помощь.
Закоментировал эти строки, чтобы не возникало проблем.

closed 2021-05-25 by Sergey1560