Sergey1560/Marlin_FB4S #44: [BUG] Медленный WiFi

MasteHiggins, 2021-07-21:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

ОЧЕНЬ медленно передаются файлы на печать из Cura. В родную прошивку всё заливается гораздо веселее.

### Version of Marlin Firmware

2,0,7,2

### Your Slicer

Cura

FialDrag, 2021-07-24:
У меня вообще при заливке через вай-фай с этой версией 2.0.9.1 СД-карта отваливается в 95% случаев, хотя с самой карты (уже со скинутым файлом) печатает без проблем. Вернулся на 2.0.7.2, там все работает отлично.

3DAngel10, 2021-08-02:
In this version (2.0.9.1) the download of the files (wifi via Cura) stops and the screen goes black. Pulling out the SD card shows you again. Another bug is that the cooling fan does not start on power up for a few ms as in version 2.0.8.1

Sergey1560, 2021-08-02:
@MasteHiggins @FialDrag  в 2.0.9 сменился слой абстракции с libmaple на HAL от ST. После этого появились проблемы с передачей файлов. На данный момент не понятна причина проблем и готового решения нет. Используйте 2.0.7.2, она наиболее стабильная из последних версий. Ничего принципиального нового в 2.0.9 нет.

@3DAngel10 Since version 2.0.9 there are problems with file transfer. The problem is not solved yet, use 2.0.7.2 as the most stable one. 
What's the problem with the fans not jerking when turned on? Why do you need this?

3DAngel10, 2021-08-02:
#define FAN_KICKSTART_TIME 100 in version 2.0.8.1 works and in this new one no, it's just to let you know (although you will surely know about it) not that I need it. Anyway you are doing a great job congratulations :)

MasteHiggins, 2021-08-03:
@Sergey1560 да, 2.0.7.2 поинтересней. Немного криво срабатывают нижние стрелки перемещения по меню (подозреваю, что по этой причине в 2.0.8.1 и была добавлена калибровка), но в целом - ОК.
Спасибо большое!

Sergey1560, 2021-10-06:
Все проблемы с картой тут https://github.com/Sergey1560/Marlin_FB4S/issues/49

closed 2021-10-06 by Sergey1560