Sergey1560/Marlin_FB4S #49: после установки последней версии не видит прошивку на карте

ginpigin, 2021-10-05:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

после установки последней версии не видит прошивку на карте, пробовал все прошивки. так же перестал видеть карту на 128ггб с любой файловой системой, фат32 тоже sd init fail. настройки не сбрасываются на дефолтные при инициализации ипром (калибровка экрана не появляется). я хотел поставить клипер и как я этт теперь сделаю?

### Version of Marlin Firmware

2.0.9.1

closed 2021-10-06 by ginpigin

Sergey1560, 2021-10-06:
При старте платы, сначала запускается bootloader, прошитый производителем на заводе. Этот бутлоадер подключает sd карту, ищет на ней файл с обновлением основной прошивки и если находит, то обновляет её. После этого просто запускает основную прошивку. Если обновления на карте нет, то bootloader просто запускает основную прошивку.
При обновлении прошивки код bootloader-а не изменяется. Он остаётся ровно тот же, что был записан производителем. Поэтому все проблемы с прошивкой, они никак не зависят от версии основной прошивки.

ginpigin, 2021-10-06:
> При старте платы, сначала запускается bootloader, прошитый производителем на заводе. Этот бутлоадер подключает sd карту, ищет на ней файл с обновлением основной прошивки и если находит, то обновляет [...]

тогда какое объяснение этому может быть? 2 карты перестали нормально работать, хотя ошибок нет, 1 карту принтер видел и печатал с неё. на пк работают обе. карта на 128ггб раньше определялась даже в exfat

aleksey-suprun, 2021-10-07:
Кстати, в репозитории марлина соответсвующий баг репорт недавно создали: https://github.com/MarlinFirmware/Marlin/issues/22898

aleksey-suprun, 2021-10-09:
Проблема решилась заменой SD карты на другую. Форматирование предыдущей в FAT или extFAT не дало никаких результатов, но с другой карточкой всё в порядке. Прошивка стартует даже с картой в слоте.