Sergey1560/Marlin_FB4S #53: 2.0.9.2 Проблемы с печатью по WiFi

PingWIn87, 2021-11-03:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Прошил принтер на marlin 2.0.9.2, брал бинарник из релиза.  
Теперь если пытаюсь отправить задание на печать из Cura, то сталкиваюсь с одним из вариантов поведения.
1. Во время отправки задания на печать в Cura  **включена** галка "Начать печать по окончании загрузки". В этом случае экран гаснет и не реагирует на внешние раздражители, печать не начинается. Лечится только перезагрузкой.
1. Во время отправки задания на печать в Cura  **выключена** галка "Начать печать по окончании загрузки". В этом случае файл загружается на карту, но карта становится недоступна из прошивки. Принтер выдаёт "No Media". Лечится извлечением карты и повторной установкой.

### Version of Marlin Firmware

2.0.9.2

### Printer model

Flyingbear ghost 5

nikitakiselev, 2021-11-07:
Тоже проблемы с флешкой. Принтер не загружается с вставленной флешкой, показывает бутлоадер, издаёт один звуковой сигнал и чёрный экран. Как только вынимаю флэшку, он снова показывает бутлоадер и загружается. 

При печати по wifi тоже проблемы. Иногда при передаче данных чёрный экран, иногда начинает издавать короткие звуковые сигналы. В итоге файл не передаётся, на флешке “error.gcode”. 

Сегодня попробую поменять флешку, может с комплектной что-то не так

Sergey1560, 2021-11-14:
По поводу проблем с картой - https://github.com/Sergey1560/Marlin_FB4S/issues/49

PingWIn87, 2022-01-16:
А по поводу "no media" при загрузке файла по wifi? Мне как-то даже нагуглить ничего не удаётся. Если нужна какая-то дополнительная диагностическая информация, то я готов её предоставить, скажите как снять.

00svd00, 2022-01-16:
Заметил что при выгрузке файла по WIFi если в этот момент будет идти активный сетевой обмен (с любым устройством, подключённым по Wifi к той же точке доступа) - после загрузки с высокой вероятностью будет Media Init Fail. Если траффика нет - почти наверняка всё будет ОК. 
При этом есть ещё такой спецэффект, что если прожата галка печати после загрузки - в ряде случаев принтер покажет BSOD и заглохнет, а в ряде случаев печать начнётся, но не запустится нагрев хотэнда либо нагрев стола. И ещё иногда при загрузке файла принтер издаёт три коротких свистка как будет произошла ошибка загрузки, но файл загружается корректно и печать проходит успешно(если руками выставить температуру стола и хотэнда) Замена карты на новую(свежекупленный SanDisk на 16Гб) также не решает проблемы.

И ещё, если прошить вот эту вебморду для MKS-WIFI модуля
https://www.mischianti.org/2021/11/28/mks-wifi-for-makerbase-robin-firmware-upgrade-and-new-web-socket-features-4/
И попытаться через него удалить любой файл с карты памяти - тоже будет мгновенный и гарантированный Media Init Fail, но файл при этом с карты удалится. И если после этого попытаться что-то запустить на печать - опять же гарантированный BSOD. Если же попытаться запустить что-то на печать до сбоя через его интерфейс - тоже будут проблемы с нагревом стола и хотэенда.

exotsk, 2022-01-17:
> Заметил что при выгрузке файла по WIFi если в этот момент будет идти активный сетевой обмен (с любым 

Все подверждаю, именно так себя оно и ведет. Печатаю в основном через малину с октопринтом.

cydwarf, 2022-01-18:
Вот всё точно так же было. Грешил на версию прошивки. Решилось заменой SD карты на допотопную с 2ГБ, пока не куплю новую (или пока не найду мелкую, огромную на 128ГБ жалко ставить в принтер). Та, что шла с принтером, показала "The media is likely to be defective" в h2testw. Заменил и ошибка ушла.

00svd00, 2022-01-18:
Карта проверена-перепроверена и даже в профилактических целях форматирована с перезаписью нулями всей поверхности. Плюс куплена новая нулёвая приличной фирмы. Там если по коду посмотреть происходит следующее - карта отмонтируется от марлина и затем через отдельную либу fatfs монтируется в отдельном процессе, который может записать файл, удалить и.т.п. При этом при записи файла там всё по полной идёт - т.е. перемонтирование карты, открытие файла на запись, запись в файл, закрытие файла, проверка записанныйх байт, перемонтирование карты обратно. Операции с ФС в этом случае проводятся через функция f_write либы fatfs и всё заканчивается хорошо, карта потом инициализируется. Если же файл удаляется то там весь процесс гораздо короче, просто перемонтируется карта, затем f_unlink(тоже из либы fatfs) и дальше перемонтирование карты обратно. И после этого стабильно всегда media init fail. Добавление задержек в разных местах на результат не влияет. Выглядит что либа fatfs как-то косячит файловую систему что марлин потом не может её переинициализировать.

Sergey1560, 2022-01-18:
Все проблемы работы с картой, связаны с переходом в Marlin на HAL от ST.
FATFs используется, поскольку драйвер фат в марлине не поддерживает запись файлов с длинными именами (в 2022 году). У основных разработчиков марлина какой-то там свой взгляд на это. FATFs по сравнению с тем, что использовано внутри марлина - эталон стабильности и надежности.

О проблеме разработчикам марлина сообщили https://github.com/MarlinFirmware/Marlin/issues/22898, но они не особо спешат исправлять и разбираться, видимо есть более важные задачи.

Исправить у себя я это не могу по 2 причинам:
1. У меня проблема не воспроизводится. У меня есть несколько карт, на них проблема не возникает. Я когда-то поймал в отладчике момент, когда карта не смогла инициализироваться и хотя бы увидел на каком этапе это происходит, но повторить не удается. Соответственно не понятно, что исправлять.
2. Код в котором проблема, он в сторонней библиотеке. Его нет в исходниках марлина. При сборке platformio автоматически скачивает эту библиотеку (ststm32) нужной версии. Исправлять нужно в ней.

Варианты решения:
1. Возможно проблемы исправлены в более свежей версии ststm32 (используется 12, а актуальная 15). Но просто изменить номер версии не получится. Остается подождать, когда тот, кто придумал это сделать таким образом (через эту ststm32) обновит код для поддержки свежей версии
2. Использовать libmaple, вместо hal от st, как это было до 2.0.8 Возможно нужно будет немного поправить код работы с wifi, т.к. не будет cmsis с объявлением регистров.

Как итог: дело не в картах, а в кривой инициализации карты марлином. Когда я видел это у себя, проблема была на этапе инициализации самой карты, еще до файловой системы.

PingWIn87, 2022-01-18:
> 1. У меня проблема не воспроизводится. У меня есть несколько карт, на них проблема не возникает. Я когда-то поймал в отладчике момент, когда карта не смогла инициализироваться и хотя бы увидел на каком этапе это происходит, но повторить не удается. Соответственно не понятно, что исправлять.

Могу поделиться картой, на которой воспроизводится.

> * Возможно проблемы исправлены в более свежей версии ststm32 (используется 12, а актуальная 15). Но просто изменить номер версии не получится. Остается подождать, когда тот, кто придумал это сделать таким образом (через эту ststm32) обновит код для поддержки свежей версии
> * Использовать libmaple, вместо hal от st, как это было до 2.0.8 Возможно нужно будет немного поправить код работы с wifi, т.к. не будет cmsis с объявлением регистров.

Я правильно понимаю, что простым пользователям остаётся просто ждать?

00svd00, 2022-01-18:
> У меня проблема не воспроизводится. У меня есть несколько карт, на них проблема не возникает. 
Стабильно воспроизводится при удалении файла на любой карте. Прям на любой карте, всегда и даже на другом принтере.

Sergey1560, 2022-01-18:
> 

Я проверю удаление.

Sergey1560, 2022-01-28:
> Стабильно воспроизводится при удалении файла на любой карте. Прям на любой карте, всегда и даже на другом принтере

Я исправил удаление. По крайней мере не должно зависать.

00svd00, 2022-02-01:
> Я исправил удаление. По крайней мере не должно зависать.

Проверил, работает. Действительно не удаляет, действительно больше не зависает. Но есть новые приколы.
1. Если передёрнуть карту - будет всегда media init failed. Стабильно. 
2. Если выгружать через вебморду mks wifi - в конец файла будут залетать блок g-кода
M105
M997
M994
M992
M27
Повторённый несколько раз(видимо в последней буфер залетает мусор из последовательного порта).
Возможно также это было всегда, я не проверял и возможно в этом одна из причин нестабильной работый. Если грузить через куру через плагин mks wifi - всё ок. Возможно также что это всё не бага, а фича)
3. Проверка в коде if( (file_size == file_inc_size) && (file_size == file_size_writen) ) фактически не работает. Если раньше она иногда рапортавала об успешной передаче файла, то теперь всегда стабильно failed. При этом если передавать файл сразу после загрузки, то несмотря на выстрел сообщения об ошибке переименования файла в file_failed.gcode не происходит(не смотря на то, что это один и тот же блок кода. Я вот этого не понимаю почему так происходит).
Если же предварительно поиграть с удалением файла и потом попытаться выгрузить файл - будет ошибка и переименование в failed. 
Мусорные блоки при этом присутствуют вне зависимости от того переименовался файл или нет(при загрузке через веб морду)
И во всех случаях файл оказывается полностью нормально переданным(опять же не считая мусорных блоков)

sma63, 2022-02-11:
Если отправлять файл на FBG5 командой:
$ curl --limit-rate 25K -v -H "Content-Type:application/octet-stream" http://192.168.0.100/upload?X-Filename=bbb.gcode --data-binary @aaa.gcode
то система не вешается, только SD-карта отсоединяется. Если потом вытащить карту и снова вставить можно продолжать работать.
Пробовал команды (освободить SD и инициировать):
$ echo "M22" | nc 192.168.0.100 8080
$ echo "M21" | nc 192.168.0.100 8080
Система вешается после M21.(((

zelenoviy, 2022-02-19:
Хм... как оказалось пару коммитов я упустил, сейчас обновился, и о чудо, передача с куры заработала! 

Теперь так же
> 1. Если передёрнуть карту - будет всегда media init failed. Стабильно.

если после этого выполнить M21 - карта цепляется, и работает, но на экране остается media init failed

Zayac316, 2022-04-15:
14.04.2022 проблема еще актуальна, хотя библиотеки вроде обновились, видимо "там" еще не вывезли=(
Есть ли у уважаемого Сергея еще старый релиз с рабочим вай фаем? ну несерьезно так каждый раз принтер дергать(

Alextesla, 2022-07-15:
На 15.07.2022 лично у меня при проблема так же осталась, по крайней мере не нашёл как решать. Но помог способ описанный выше. Нашёл древнюю карточку на 2гб, времен когда 2гб ещё считалось круто и много, без ничего вместе со всеми файлами вставил в принтер и - о чудо - печать заработала как и была на стоке, отправляешь файл из куры, все грузится, все настройки сохраняются и принтер все печает, по крайней мере все видимые настройки переносятся из куры в принтер. Жаль только до этого уже купил новую флешку за косарь(

andreypopov, 2022-12-24:
> Та же проблема с отправкой заданий с Cura. По окончанию отправки - пищит, экран выключается. После перезагрузки файл лежит на карте, можно печатать с нее. Воспроизводимость 100%, карта родная с принтера, 4Gb Прицепился терминалом по usb, в момент ошибки выдается Error: volume.init failed Судя по коду - происходит в момент вызова card.mount() в mks_wifi_sd_deinit()
> 
> Для эксперименту воткнул в mks_wifi_sd_init() (вызывается перед операциями с файлами- загрузка/удаление) такой кусок
> 
> // отцепляем от марлина, монтируем и отмонтируем в FATFS, цепляем обратно к марлину, работает SERIAL_ECHO_MSG("X 1"); card.release(); res = f_mount((FATFS *)&FATFS_Obj, "0", 1); f_mount(0, "", 1); card.mount();
> 
> // то же самое с открытием файла на чтение через FATFS, работает SERIAL_ECHO_MSG("X 2"); card.release(); res = f_mount((FATFS *)&FATFS_Obj, "0", 1); res = f_open((FIL *)&upload_file,"test",FA_READ); f_close((FIL *)&upload_file); f_mount(0, "", 1); card.mount();
> 
> // то же самое но на запись(создание) файла, файл на карте создается, НО ВСЕ, тут марлин обратно не может примаунтить карту ! SERIAL_ECHO_MSG("X 3"); card.release(); res = f_mount((FATFS *)&FATFS_Obj, "0", 1); res = f_open((FIL *)&upload_file,"test2",FA_CREATE_ALWAYS | FA_WRITE); f_close((FIL *)&upload_file); f_mount(0, "", 1); card.mount();
> 
> Насколько понял, в fatfs своя реализация sdio (sdio_driver_*), у марлина своя (src/HAL/STM32(STM32F1??)/sdio.cpp). Функция card.release() фактически ничего не делает с устройством/драйвером, только проставляем у себя (cardreader) флаг монтирования. Не может ли быть такого что после работы fatfs состояние устройства изменилось, но при монтировании обратно к марлину его драйвер то этого не знает, храня устаревшее состояние ? При ошибке на card.mount() вызов driver->init() происходит успешно(возможно инициализация заново и не происходит) , а падает на volume.init() при попытке ReadBlock с устройства На этом мои познания в МК иссякают, надеюсь записки сумасшедшего могут в чем то помочь =)

точно такая же ошибка, прошивка последняя из релизовов (август 2022)
как подлечить, очень сильно надоело

Zayac316, 2022-12-26:
> точно такая же ошибка, прошивка последняя из релизовов (август 2022) как подлечить, очень сильно надоело

Странно, у меня на последней версии прошивки (2.1.1) все работает прекрасно.