Sergey1560/Marlin_FB4S #55: [FR] Настройка linear advance на экране печати

feivur, 2021-12-05:
### Describe the feature you want

Хочется иметь возмжность менять linear advance во время печати, как и параметры температуры, например.

Sergey1560, 2021-12-11:
Такая возможность есть. Во время печати нажмите символ шестеренки, откроется меню настроек. Там есть возможность изменить LA.

closed 2021-12-11 by Sergey1560

feivur, 2021-12-12:
[image]

feivur, 2021-12-12:
Такой возможности нет(.

feivur, 2021-12-12:
[image]

Sergey1560, 2021-12-12:
Не используйте интерфейс от MKS. В классическом COLOR UI эта возможность есть.

feivur, 2021-12-12:
Я бы рад от него отказаться. Но в классическом плагин в куре не подключается к принтеру по wifi. Пробовал компилировать с и без флага MKS_WIFI_MODULE.