Sergey1560/Marlin_FB4S #56: Запрос замены филамента

exxico, 2021-12-11:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

При печати несколько раз (минимум два) принтер останавливается, выплёвывает филамент и ждёт замены. слайсер - кура с дефаультными настройками. Печать моделей с thingiverse.com

### Version of Marlin Firmware

2_0_9_2

### Printer model

FlyBear5

### Electronics

robin nano v1.2

### Bed Leveling

ABL Linear grid

### Host Software

Cura

### Additional information & file uploads

напрягает.

akarpik, 2021-12-11:
Please check Runout Filament Sensor. In most cases loosen connection of the cable could lead to described error.

Sergey1560, 2021-12-11:
Проблема в самом датчике наличия филамента. Если вы не предполагаете, что филамент может кончится, просто отключите его в меню.

exxico, 2021-12-12:
Воот! в меню на принтере не нашёл такой опции. датчик в порядке - отрабатывает наличие прутка. даже светом индицирует. ткните носом, плиз, где в настройках отрубить такое поведение?

Sergey1560, 2021-12-22:
В меню принтера это Runout sensor. Там можно его выключить.
В Marlin/Configuration.h - #define FILAMENT_RUNOUT_SENSOR и дальше его параметры.
По-умолчанию этот датчик выключен (#define FIL_RUNOUT_ENABLED_DEFAULT false).

Если вы используете MKS UI, там обработка датчика филамента внутри этого интерфейса. Настройки датчика в Marlin/src/pins/stm32f1/pins_MKS_ROBIN_NANO_common.h  Если заданы MT_DET_1_PIN или MT_DET_2_PIN, то на них будет проверяться уровень. Если уровень MT_DET_PIN_STATE, то считается, что датчик сработал. Чтобы отключить его, достаточно закомментировать MT_DET_1_PIN и MT_DET_2_PIN

closed 2022-01-18 by Sergey1560