Sergey1560/Marlin_FB4S #57: Питание на пине CS

Esmertec, 2021-12-15:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Не совсем баг, но все же решил апнуться с прошивки 2.0.7.2 до 2.0.9.2 разумеется. Проблема в том, что у меня стоял модуль автоотключения через релюшку подцепленную на CS = PE5. На 2.0.9.2 принтер вырубается т.к. на этом пине (CS = PE5) нет напряжения. Как его включить обратно? Заранее спасибо за ответ!

### Version of Marlin Firmware

2.0.9.2

### Printer model

FBG5

Sergey1560, 2021-12-22:
Вам нужно указать SUICIDE_PIN. В 2.0.9 немного иная структура в файлах pins*, общие ноги для robin nano плат вынесли в отдельный файл Marlin/src/pins/stm32f1/pins_MKS_ROBIN_NANO_common.h
Укажите в файле для вашей платы (например Marlin/src/pins/stm32f1/pins_MKS_ROBIN_NANO.h если плата robin nano v1.1) настройки SUICIDE_PIN. Их можно скопировать из 2.0.7

Esmertec, 2021-12-22:
Спасибо за ответ, попробую сделать. По поводу новой структуры я уже понял, т.к. начиная с 2.0.8.1 пины не работают - проверено xD Отпишусь по результату

closed 2022-01-18 by Sergey1560