Sergey1560/Marlin_FB4S #58: Релиз 2.0.9.2 для 5й версии не загружается

VyacheslavTeplyakov, 2021-12-26:
Перепробовал версии для всех комплектов драйверов, по идее это вообще влиять на загрузку не должно, но попробовал и fb_5 и fb5_all_drv_2208.
Плата красная, значит 1,2 и по расположению элементов и надписям совпадает с той что на картинке.
Вроде как прошивается, потом показывает логотип марлина, пикает один раз и всё, темно серый экран с подсветкой и никаких дальнейших признаков жизни. На сток вернулся как ни в чем ни бывало. Что не так с экраном?

Sergey1560, 2021-12-27:
С экраном скорее всего все так, проблема с картой. Подробности тут https://github.com/Sergey1560/Marlin_FB4S/issues/49

VyacheslavTeplyakov, 2021-12-27:
> С экраном скорее всего все так, проблема с картой. Подробности тут #49

да, в итоге карта косячная была, заменил на нормальную брендовую и все взлетело. Спасибо!

closed 2021-12-27 by VyacheslavTeplyakov