Sergey1560/Marlin_FB4S #59: Плата Flyingbear Reborn v3.0

mugabe, 2022-01-24:
FB5 пришёл с такой платой, по компоновке очень похожа на Robin Nano V3.0, но с распаянными драйверами и STM32F**103**.
Подойдёт ли к ней какая-то прошивка?

[image]
[image]

Sergey1560, 2022-01-24:
Эта плата больше всего похожа на Robin Nano-s 1.3, только с stm32f103, вместо stm32f407. В схеме Robin Nano-s 1.3 обозначена возможность установки как f407, так и f103.
К этой плате, скорее всего, подойдут прошивки для f103. Единственное, что нужно будет изменить - направления осей. Я думаю, подойдет набор от Robin Nano-s 1.3.

Sergey1560, 2022-01-28:
Для этой платы подойдет сборка для F103 (mks_robin_nano35 или mks_robin_nano35_maple в platformio)
В configuration.h добавлен набор направлений для осей FB_5_REBORN_3_0

Al3eRtO, 2022-11-29:
Hi, 
so the right firmware is mks robin nano v1 ( Robin Nano-s 1.3) or v2 (mks_robin_nano35) ?