Sergey1560/Marlin_FB4S #60: FB5 Прошивка не запускается со вставленной CD картой.

Tonic04, 2022-01-27:
Доброго времени. Год не обновлял прошивку, тут, решил заняться. 
Для начала хотел посмотреть что нового в последнем Marlin_FB4S-MKS_UI, но собрав прошивку на платы v. 1.1. - 1.2. обнаружил некоторые нововведения:
1. Прошивка заливается, потом темный экран до момента, пока не извлеку CD карту. После извлечения прошивка запускается дальше.
2. При запуске прошивки не подгружаются картинки меню и на них нет данных. Что только не предпринимал, попыток 10 точно перезалива с разными параметрами, даже под v.1.3. (вообще не запустилась).
3. Ставлю прошивку из основной ветки - опять требует извлечения CD карты, причем каждый раз при включении принтера, иначе не стартует. Артефакты прошивки с карты удалял, но ничего не помогло. 
Не могу понять что не так делаю. Возможно плохо использовал поиск, но инфы не нашел. Одни невидимки CD. У меня как-то наоборот очень хорошо видит, аж не запускается.

Sergey1560, 2022-01-28:
Проблемы с картой памяти описаны тут https://github.com/Sergey1560/Marlin_FB4S/issues/49
Как одно из решений на данный момент, попробуйте в качестве default_envs в platformio.ini указать mks_robin_nano35_maple. Это позволит собрать прошивку с libmaple, вместо HAL.

По поводу картинок: на карту памяти при прошивке нужно скопировать папку assets, находится там же, где файл с прошивкой. При старте картинки записываются во внутреннюю память и этот процесс отображается на экране.

С MKS_UI есть [проблема со сдвигом слоев по оси Y](https://github.com/MarlinFirmware/Marlin/issues/22967) При сборке с libmaple этой проблемы нет.

Tonic04, 2022-01-28:
Благодарю за ответ. Да, действительно, по части MKS_UI внимательно прочел на буржуйском, assets скопировал, картинки появились.
По поводу CD сходил, купил новую на 16, все стало с ней работать. Старая 4 ГБ форматировалась вчера 5 часов, к сожалению другой на тот момент не было.

В итоге вернулся к обычному цветному Marlin из основной ветки. MKS_UI что-то не понравился дизайном, какой-то он серо-бармалиновый. Мне нравится дизайн родной прошивки, но не нравится отсутствие возможности нормально перевести и сделать пункты меню заглавными буквами.

pash7ka, 2022-04-09:
Подтверждаю проблему с MKS_UI. У меня FB4s c драйверами TMC2209 UART, на дефолтном `mks_robin_nano35` подтверждаю случайные сдвиги оси Y. На COLOR_UI с точно такими же настройками их нету, как и при изменении _default_envs_ на `mks_robin_nano35_maple`.

@Sergey1560 Может быть стоит вынести данную информацию в README? Я бы ни за что не догадался проверить, если бы мне в ТГ не рассказали про этот баг.