Sergey1560/Marlin_FB4S #62: Marlin 2.0.9.3 fb4s_All_TMC2208

LieToMe51, 2022-02-03:
Хелп! Перестаёт работать мотор экструдера после 3го слоя - где-то 30 минут печати.
До этого печатал маленькую модель 8-10 минут и все было нормально.
Драйверы настроены на: x - 1.2В; y - 1.2В; z - 1.2В; e - 1.4В. 
[image]
В чем может быть причина?!

nixprosoft, 2022-02-04:
Я думаю, что если Вы приложите G-CODE который крашится, то диагностировать будет проще.

Sergey1560, 2022-02-07:
Добрый день.
LA используете? С 2208 и 2225 чаще всего проблема в использовании Linear Advance.  Описание проблемы от TMC есть в issue Klipper - https://github.com/Klipper3d/klipper/issues/196#issuecomment-383122655
Если проблема в этом, то можно или переключить драйвер в spreadcycle или использовать 2209.

closed 2022-02-14 by Sergey1560