Sergey1560/Marlin_FB4S #63: Marlin 2.0.9.3 MKS_UI сходит с ума после 30 минут печати

baralgin1003, 2022-02-06:
около 8-10 слоев печатает нормально, но потом голова начинает ходить вхолостую и происходит смещение координат по Y около 1см, далее минут 5 продолжается печать "по воздуху" и принтер с писком уходит в перезагрузку... EEPROM после прошивки инитил, конфиги не трогал

Sergey1560, 2022-02-07:
Добрый день.
Проблема смещения координат при использовании MKS UI присутствует в оригинальном коде Marlin. Описание проблемы - https://github.com/MarlinFirmware/Marlin/issues/22967
В качестве способа решения, попробуйте использовать в качестве default_envs в platformio.ini mks_robin_nano35_maple

baralgin1003, 2022-02-07:
Спасибо за быстрый ответ! Сборка велась как раз через mks_robin_nano35_maple. Видимо, придется обойтись обычным Марлиновским интерфесом.

pash7ka, 2022-04-09:
Подтверждаю проблему с MKS_UI. У меня FB4s c драйверами TMC2209 UART, на дефолтном `mks_robin_nano35` подтверждаю случайные сдвиги оси Y. На COLOR_UI с точно такими же настройками их нету, как и при изменении _default_envs_ на `mks_robin_nano35_maple`.

@Sergey1560 Может быть стоит вынести данную информацию в README? Я бы ни за что не догадался проверить, если бы мне в ТГ не рассказали про этот баг.

Sergey1560, 2022-04-11:
> Подтверждаю проблему с MKS_UI. У меня FB4s c драйверами TMC2209 UART, на дефолтном `mks_robin_nano35` подтверждаю случайные сдвиги оси Y. На COLOR_UI с точно такими же настройками их нету, как и при [...]

В целом можно конечно описать в readme. Тем более, что в марлине не очень шевелятся что-то делать по этому поводу. 
Но раздел issue, он как раз про проблемы :) Собственно в открытых issue тут описаны известные проблемы, которые пока не имеют полноценного решения.

pash7ka, 2022-04-11:
Да, понимаю про раздел issue, но мне было вообще сложно догадаться, что источником проблемы может быть не механика или настройки тока, а баг прошивки.
А тут по факту получается, что выполнив инструкцию из README ветки MKS_UI я получаю неработающий принтер. И это несколько не то, что ожидалось.