Sergey1560/Marlin_FB4S #64: Marlin 2.0.9.3 & FAST_PWM_FAN & Filament sensor

kirichenkomaxim, 2022-02-11:
FBG5, плата Robin Nano v1.2
Из-за странной(писк, поздний старт) работы кулеров охлаждения был активирован FAST_PWM_FAN
в частности в Marlin\Configuration_adv.h :

#define FAN_MIN_PWM 62
#define FAN_MAX_PWM 255
#define FAST_PWM_FAN
#define FAST_PWM_FAN_FREQUENCY 39200

FAST_PWM_FAN_FREQUENCY подобрана опытным путем, кулеры стали работать корректно(стартуют от 5%, не шумят и не пищат). Но...проявилась проблема: через 1-2 минуты печати начинается "смена филамента", после замены все повторяется; если отключить (в меню) датчик филамента - проблема не проявляется.
Пробовал использовать FAN_SOFT_PWM - проблема с произвольной сменой филамента отсутствует, но кулеры начинаю потрескивать и стартуют уже только с 30%

P.S. Все рекомендации по прошивка выполнены не единожды, EEPROM зачищен;

Sergey1560, 2022-02-14:
Я полагаю, проблема где-то в железе. Не вижу, что тут можно "чинить" программно. 
С датчиком филамента, проверьте что он подтянут в настройках к нужному уровню. У FB5 и FB4S датчики с разным активным уровнем и это часто вызывает путаницу.

kirichenkomaxim, 2022-02-14:
Дело в том, что если не делать "тюнинг" для кулеров(вышеописанный FAST_PWM_FAN), то проблемы с датчиком не наблюдается - печать проходит без прерываний.

Noctr, 2022-07-11:
"... У FB5 и FB4S датчики с разным активным уровнем и это часто вызывает путаницу ..."
По моему стоило бы указать, для тех, кто собирает прошивку, что необходимо выбрать в
.../Marlin/src/pins/stm32f1/pins_MKS_ROBIN_NANO.h
49   //#define PRINTER_NAME_FB5
50   #define PRINTER_NAME_FB4S
нужный тип принтера.

Davilka, 2022-07-13:
> "... У FB5 и FB4S датчики с разным активным уровнем и это часто вызывает путаницу ..." По моему стоило бы указать, для тех, кто собирает прошивку, что для корректной работы датчика тип принтера выбирается в .../Marlin/src/pins/stm32f1/pins_MKS_ROBIN_NANO.h 49 //#define PRINTER_NAME_FB5 50 #define PRINTER_NAME_FB4S

Мне кажется, это надо просто вывести не в pins а в Configuration.h