Sergey1560/Marlin_FB4S #69: Не корректная работа защиты на FBG5 robin nano v1.1.

RayCustom, 2022-05-29:
Медведь ни как не хочет печатать дольше часа. 
Начинает пищать, пишет "Е1 max temp" а иногда ничего не пишет, и отключается, установлено твердотельное реле. Что интересно в новой прошивке fbg5_all_2208 нет сигнала на управляющем Пине для реле, это у меня что-то не так?
Проблема с произвольным отключением не решается заменой термо датчика, нагревательного блока. Подскажите что можно сделать?

Davilka, 2022-07-13:
> Медведь ни как не хочет печатать дольше часа. Начинает издавать звук, пишет "Е1 max temp" а иногда ничего не пишет, и отключается, установлено твердотельное реле. Что интересно в новой прошивке fbg5_all_2208 нет сигнала на управляющем Пине для реле, это у меня что-то не так? Проблема с произвольным отключением не решается заменой термо датчика, нагревательного блока. Подскажите что можно сделать?

У меня принтер произвольно перезагружался через 15-30 минут после начала печати.
Помогло отключение WATCHDOG:
1957 // #define USE_WATCHDOG

Zayac316, 2022-07-26:
Привет товарищи, немного другая проблема, даже при установке в конфигурации MAXTEMP на экструдер в 400 градусов, не дает греть выше 280 градусов, я так понимаю это стандартная температура которая была изначально в прошивке + погрешность перегрева.
как выключить это?