Sergey1560/Marlin_FB4S #72: 2.1.1 Reborn v3.0 зависает на "TFT updating 100%".

AtomHell, 2022-08-13:
Расширение прошивки на карте памяти меняется на *.cur . Карту форматировал SD Card Formatter'ом. Родная прошивка с этой же карты встаёт без проблем. Пишу в первый раз, сильно не пинайте, если что не так.

10on, 2022-09-05:
Присоединяюсь,  проблема аналогичная.
Откат на предыдущий релиз проходит успешно. 

Если попробовать перезагрузить принтер - экран не работает, репитер хост подключаться не хочет.
Звук включения со спикера есть.

AtomHell, 2022-09-05:
Пробовал собирать из исходников 2.1.1 mks_robin_nano35 и mks_robin_nano35_maple. И всё равно прошивка не встаёт.

Sergey1560, 2022-09-09:
Судя по всему, я прошивку для Reborn v3.0 собрал под F4. Собранная под F1 нормально запускается, но есть проблемы с подключением карты памяти. Эта проблема не только на Reborn v3.0.

closed 2022-09-14 by AtomHell