Sergey1560/Marlin_FB4S #74: Долгий нагрев с последующим отключением подогрева.

Ratevandr, 2022-10-22:
После начала печати идет долгий нагрев (от 5 до 10 минут, числа достигают нужной температуры и колеблятся около неё), после чего целевая температура стола и сопла обращается в 0. Печать при этом продолжается (но филамент, само собой не идет, сопло стремительно остывает). 
Воспроизводится раз через раз, не знаю с чем связано.
Прошивка: fb5_all_drv_2208 версии 2.1.1
Пример gcode
[Originale.txt](https://github.com/Sergey1560/Marlin_FB4S/files/9844985/Originale.txt)

cydwarf, 2022-12-03:
Да, иногда бесит это обнуление. Приходится следить за экраном, пока не начался первый слой.

> колеблятся около неё
Колебания температуры я решил PID под каждый пластик.

certus86, 2023-11-12:
Та же проблема. Алгоритм пока ещё не выявил.
Меняю пруток. Хотенд нагрет. Запускаю печать. Греется стол. Стол нагревается, запускается нагрев хотенда и стол сбрасывается в ноль. Начинается печать, а стол остывает