Sergey1560/Marlin_FB4S #75: Ghost 5 with reborn v 3.0 motherboard is not a bootable on 2.1+ versions

dmitrychepel, 2022-11-09:
Now I have 2.0.9.3_1 version from _releases_ on my ghost 5.
I tryed update to new 2.1.1 version from _releases_ for LA fixes. and I see that TFT update finished (become to 100%) and freeze. after that ghost become to unbootable state. I could revert to 2.0.9.3 but 2.1.1 does not work.
I tryed to change sdcard, tryed to boot without sdcard, and it is not help me.
BTW on the 2.0.9.3 works fine both versions libmaple and st_hal.

Sergey1560, 2022-11-12:
Судя по всему, сборка для платы reborn v3.0 сделана не под тот тип процессора в бинарниках для 2.1.1

dmitrychepel, 2022-11-12:
Если что надо попробовать, то могу.

ConstantinoGoo, 2022-11-30:
> Если что надо попробовать, то могу.

Пробовали обновленную прошивку 2.1.1?

dmitrychepel, 2022-12-01:
как я понял сообщение удалили, не смог его найти но если что версию 2.1.1 августовскую пробовал. более новую, если такая есть, не вижу.

ConstantinoGoo, 2022-12-01:
> как я понял сообщение удалили, не смог его найти. Но если что версию 2.1.1 августовскую пробовал. более новую, если такая есть, не вижу.

Да, я почему-то подумал, что ставил 2.0.9..., а висит 2.1.1. А потом как понял... Извиняюсь за беспокойство.

ConstantinoGoo, 2023-01-02:
2.1.2 works fine with FB Ghost 5.

dmitrychepel, 2023-01-07:
Теперь Реборн v3 загружается. Эта проблема решена.

closed 2023-01-07 by dmitrychepel