Sergey1560/Marlin_FB4S #76: Ghost 5 with reborn v 3.0 motherboard is not able to start printing on 2.1.2+ versions

ConstantinoGoo, 2023-01-04:
The printer works fine until the green check mark is pressed. Then the printer just freezes

[image]

https://user-images.githubusercontent.com/49076763/210572595-f379513b-cb35-4057-89b6-7562595564e0.mov

BWorkshop, 2023-01-07:
У меня аналогичная проблема после прошивки на FBG5 с платой Robin Nano 1.2, зависает после подтверждения выбора файла, РепетирХост выводит Read error. Кроме этого, происходит остановка оси X во время печати, буквально на первом же слое, возможность движения восстанавливается после перезагрузки. На осях X и Y стоят TMC2208. Никак не получается решить.

Sergey1560, 2023-01-07:
Явно есть какие-то проблемы при работе с картой. Попробуйте готовую прошивку из bugfix. Я в последнем релизе выложил архив. Это ветка bugfix марлина с изменениями только в разделе конфигурации. Весь код ровно тот, что был, без изменений.  Не будет работать wifi, но будет понятно, проблема связана с моими изменениями под wifi модуль, или нет.

ConstantinoGoo, 2023-01-08:
> Явно есть какие-то проблемы при работе с картой. Попробуйте готовую прошивку из bugfix. Я в последнем релизе выложил архив. Это ветка bugfix марлина с изменениями только в разделе конфигурации. Весь [...]

Прошивка .../firmware-2/bugfix/stm32f103_boards/reborn_v3 в версии 2.1.2_2 прошивается и запускается на печать корректно.

BWorkshop, 2023-01-09:
После исправления Сергеем 09.01.2023 я установил прошивку из ветки FB4S_WIFI, все стало работать как нужно. SD карта читается, две печати прошли без сбоев по осям.

ConstantinoGoo, 2023-01-13:
В актуальной версии прошивки по состоянию на 13.01.2023 не актуален.

closed 2023-01-13 by ConstantinoGoo