Sergey1560/Marlin_FB4S #79: 2.1.2_3. FB5. Reborn v3.0. Остановка движения по оси X.

ConstantinoGoo, 2023-01-13:
Во время печати экструдер перестаёт передвигаться по оси X, но продолжает печатать. В итоге пластик наплавляется одной стенкой.
Запущено 3 печати, во всех 3 дефект проявился (причём +- в одном и том же месте).

[image]

Sergey1560, 2023-01-13:
Попробуйте готовую сборку из папки bugfix (в 2.1.2_2 были собраны) с этим же gcode-дом.

Sergey1560, 2023-01-16:
Попробуйте установить тип драйвера TMC2208_STANDALONE для осей, где используется 2208 или 2225. 
Я пересобрал прошивки с указанием типа драйвера.
Возможно, после добавления алгоритма IS, драйвер отключался так же, как при работе с LA. Для 2208 добавили недавно исправление, но оно работает, если указать нужный тип драйвера.

closed 2023-02-03 by ConstantinoGoo