Sergey1560/Marlin_FB4S #81: 2.1.2_4, принтер перезагружается во время печати, FBG5, Robin Nano 1.3, 2 драйвера а4988

andrmiw9, 2023-02-02:
Начнём с того, что это моё 1-ое знакомство с марлином, так что опыта 0. Принтер полностью стоковый, это его первая прошивка.

Предисловие:
Использовал файл прошивки (.bin) из папки:
mks_robin_nano_v1_3_a4988 - Robin Nano 1.3, два драйвера А4988, два драйвера TMC, в соответствии с моей конфигурацией принтера. Форматнул SD-карту из комплекта и перебросил файл, обновил прошивку, сразу же сбросил настройки принтера (Initialize EEPROM). Wifi был подключен ещё на стоковой прошивке, 3д принтеру выдал статический ip на роутере.

Прошился, сбросил настройки принтера, поставил на печать обычный 3d benchy кораблик. Файл принялся и после этого принтер тупо перезагрузился. Дальнейшие попытки - всё тот же эффект, иногда даже циклично начинал перезагружаться, а не просто один раз. Попробовал вручную закинуть файл на карту памяти, начал печать, пошёл процесс, всё нагрелось, но в рандомном месте всё остановилось. Несколько раз повторил эту процедуру, выяснилось: принтер останавливается на рандомном слое, в совершенно рандомном месте. Попробовал и другую модельку - тот же эффект. Печатающую голову при этом принтер не отводит, она так и остаётся на месте остановки. Поставил штатив, заснял момент перезагрузки на телефон - принтер ошибок не выдаёт, просто экран перестаёт обновляться (XYZ и температуры зависают) и через пару секунд перезагрузка. 

Начитался в инете про проблемы с SD-картой, пробовал её вставлять после включения принтера, форматировать, пробовал другие SD-карты, пробовал даже другие переходники для ПК. Раз 5 уже перепрошивал, используя всё тот же файл .bin. Один раз выдало e1 heating failed printer halted please reset, но не уверен, что это связано, возможно просто в комнате было прохладно, и сопло не успело нагреться (хрен его знает)

В общем и целом, поставил сейчас версию 2.1.1 и кораблик с первого раза напечатался через Wifi, пока всё стабильно хорошо.
Возможно кому-то мой опыт поможет

ZelenyjScript, 2023-02-02:
Использую сборку 2.1.2_4 7 дней.

Подтверждаю ту же ошибку для FBG5 с платой версии 1.2. Использую прошивку mks_robin_nano_v1v2_fb5_2208.
Отправляю файлы на печать из Cure по WiFi. Примерно на 10% отправленных на печать файлов, сразу после upload, происходит самоперезагрузка принтера. Файл при этом уже есть на microSD и его можно запустить на печать. Очистка microSD и повторная отправка точно того же файла снова приводит к перезагрузке.

Прочие сбои тоже бывают и очень разнообразные: саморазгон нагрева со срабатыванием защиты (1 случай), отказ экструдера по время печати (3 случая) при полной его исправности.

Скорее всего проблема не сборки, а самого Марлина.

Egorussiaa, 2023-02-09:
Это у вас товарищи еще цветочки. У меня FB5 c платой 1.3 все драйверы TMC. На версии 2.1.2_3 и _2 у меня принтер сразу ехал печатать без попыток нагрева стола и экструдера. На версии 2.1.2_4 нагрев происходит но, подтверждаю баг с тем, что при печати принтер уезжает в случайном порядке по любой оси. И ладно X Y ездит, оставляет петельки пластика на столе. Он у меня решил по Z и прожег не допечатанный кубик фильерой. 

Поставил 2.0.9 все работает пока что. 

Ну а сборщику благодарность, я бы с чистого марлина рабочую версию бы ни в жисть не собрал.

SantaLemuroff, 2023-03-09:
"У меня FB5 c платой 1.3 все драйверы TMC" - на XY 2208 или 2209?

ZelenyjScript, 2023-03-22:
Произвел некоторую модернизацию принтера и проблема остановки экструдера прекратилась. Прежде всего, поменял двигатель NEMA17 на вдвое меньший в рамках переделки на direct: с 40 мм на 20 мм, но с рабочей температурой до 180°С. Установил радиатор, обдув и снизил рабочий ток на драйвере. Без радиатора двигатель раскаляется минут за 20 печати. Больший двигатель перегревался медленнее, но и был подешевле и рассчитан был до 80°С, а потому при перегреве была остановка печати по отказу двигателя.
Marlin умудряется сильнее нагревать шаговики, чем стоковая прошивка FGB5.

andrmiw9, 2023-07-15:
Поменял ещё 2 SD-карточки, на 1 из них всё заработало, 2 печати пока успешно, будем смотреть дальше. Сохраняю gcod-ы в путь с пробелами, это на ошибку не влияет.

Alexious-sh, 2024-04-05:
2.1.2_5, плата MKS Robin Nano-s v1.3 - та же беда.
Если сразу включить принтер и запустить модель на печать - уходит в ребут через +/- минуту.
Если перед печатью подвигать оси (например, Home XY) - начнет "печатать" без нагрева стола и екструдера.