Sergey1560/Marlin_FB4S #82: При прошивке 2.1.1 и выше платы Reborn3 (впаянные 2225) начинает зависать ось Z при печати

Stonerain72, 2023-02-13:
Есть два Медведя. Один с платой Reborn v2.0, второй Reborn v3.0.
На медведе с платой Reborn2 прошивка 2.1.1 для Reborn2 работает корректно на оси Z. При попытке прошить 2.1.1 или 2.1.2 плату Reborn3 (соответствующей готовой сборкой от автора Sergey1560 или ручной сборкой - не важно) - появляется зависание оси Z.
Соответственно, прошивка Sergey1560 версией 2.0.9 или родной фирменной V72-T2225 глюк устраняет.
Но при этом теряется LA для TMC2225. Печалька. Было бы хорошо спросить Медведей - какая версия Марлин в их последней фирменной прошивке, а то может и LA пойдет =)
И что любопытно, на 2.1.1 звук оси Z у Reborn3 был резкий и как бы с треском, когда она двигалась. При переходе на 2.0.9 звук стал мягкий, правильный. При тех же скоростях-ускорениях.
Данный вопрос я поднимал тут (и сам же нашел решение): 
https://3dtoday.ru/questions/stala-otklyucatsya-os-z-na-flying-bear-ghost-5-pryamo-vo-vremya-pecati-est-versii

Sergey1560, 2023-02-13:
Что за зависание оси Z? Ось прекращает движение? Если остановить печать и отправить команды на перемещение по Z, двигается?
Решение вы уже нашли? В чем оно заключается?

Stonerain72, 2023-02-13:
Спасибо за реагирование на обращение. Да и вообще за Ваш труд =)
Я детально все описал на форуме тут: (в головном сообщении все описал, что происходит)
https://3dtoday.ru/questions/stala-otklyucatsya-os-z-na-flying-bear-ghost-5-pryamo-vo-vremya-pecati-est-versii
Если коротко, то да - есть 2 варианта поведения оси Z:
1. Запускается печать модели, голова нормально паркуется по всем осям, потом идет из предпечатного скрипта команда опускания стола на 20 мм, голова подходит к точке начала печати, а стол уже обратно не поднимается в 0. ШД при этом легкий - не запитан. И до выключения питания он уже не реагирует ни на какие команды.
2. Все то же, что и в п1, но стол нормально поднимается в 0, идет печать 1-го слоя, но затем ось Z уже не двигается. Вся печать идет в 1-й слой. Хотя координата Z на дисплее растет, то есть прошивка считает, что ось двигается.
Вылечил все откатом на Ваш 2.0.9. Тут все хорошо.

Sergey1560, 2023-02-13:
Готовую прошивку использовали из релиза 2.1.2_4 ? 
Похоже на остановку драйвера 2208/2225 из-за включенного алгоритма IS. По оси Z этого вроде не должно было быть.  Если собираете сами, проверьте, что driver type стоит tmc2208_standalone для осей на которых 2208 или 2225 (на плате Reborn 3.0 это все оси). 
В 2.1.2_4 я это исправил и в configuration.h добавлено это, для самостоятельной сборки (нужно выбрать плату в начале файла).

Stonerain72, 2023-02-13:
2.1.2-4 - да - ставил. На ней ось Z барахлит. Как и на самосборной 2.1.1.
Причем Ваша уже готовая 2.1.1 на плату Реборн-3 не установилась. Сначала пытался именно ее поставить, поскольку на Реборн-2 она работает у меня отлично. После этого поставил Вашу собранную 2.1.2-4. Не помогло. Только возврат к 2.0.9 избавил.

Stonerain72, 2023-02-15:
Сергей, здравствуйте. Кое-что прояснилось. Продолжая сопоставлять условия и факты, вот что удалось установить.

Два принтера с чуть разными платами (Reborn v2.0 и Reborn v3.0), но с одной версией прошивки (под каждую модель платы - своя сборка), работают с осью Z по-разному. Одна и та же модель на одном принтере печатается нормально, на другом Z парализуется с отключением ШД. Слайсилась модель в Прюше-слайсере. Это важно.

ВЫЯСНИЛОСЬ! =) что отчасти проблема в G-коде от Прюш-слайсера. И это крайне любопытно. Этот самый Прюш-слайсер ставит для оси Z скорость перемещения при печати F7800, то есть 130мм/сек. Но в прошивке стоит ограничение максимальной скорости по Z - максимум 10мм/сек. И теоретически (!) прошивка должна гасить указанную скорость до значения параметра DEFAULT_MAX_FEEDRATE { 120, 120, 10, 40 }

Но выходит так, что на плате Reborn v2 с ейной сборкой прошивки это отрабатывается корректно, а на плате Reborn v3 с ейной сборкой прошивки DEFAULT_MAX_FEEDRATE игнорится, и ось Z летит со скоростью 130мм/сек, что у нее, естественно не получается, и принтер блокирует ее работу.
Обнаружилось это благодаря успешной печати на Reborn v3 модели с кодом не из Прюше-слайсера. Сравнил G-коды и увидел, что в коде из другого слайсера на ось Z прописана нормальная скорость F600. И оно нормально печатается без зависания оси Z.
Таким образом, тут совмещение двух багов из двух систем: Кривость Прюше-слайсера, которая, как оказывается тянется уже 2-3 года и разработчики НЕ хотят менять это. Народ на ино-форумах давно жалуется на параметр F7800 в коде для оси Z. Но обычно тут выруливает прошивка, которая понижает скорость до указанной в прошивке. А именно на Reborn v3 прошивки, начиная с 2.1.0 не режут эту F7800 до максимально-допустимой. И ось сначала рвет спину, что я и слышал - рев и вибрации, а потом зависает, благодаря какой-то там защите принтера.

Вот примерно так. Это поправимо?

Sergey1560, 2023-02-15:
А что указано в лимитах принтера по оси Z?
Попробуйте задать небольшую скорость по Z:
M203 Z5 (тут значения в мм/с)
А потом отправить его по Z с большой скоростью:
G0 Z100 F7800

Ограничения скорости срабатывают?

Stonerain72, 2023-02-15:
В принципе, я все максимально подробно написал, в том числе и про свои настройки максимальной скорости:
DEFAULT_MAX_FEEDRATE { 120, 120, 10, 40 }. Вроде как попытался всю информацию, о которой Вы спрашиваете детально выложить.... 
На Reborn v2.0 скорость ограничивается до 10, печать идет нормально, а на Reborn v3.0 пытается переть со всеми 130.
Запустить просто Ваш код попробую (завтра на работе) . Но это точно имеет смысл, если, вобщем, оно уже косвенно проверено?

Stonerain72, 2023-03-16:
Сергей, здравствуйте еще раз. Наконец-то добрался проверить принтер по Вашему вопросу:
**********************
А что указано в лимитах принтера по оси Z?
Попробуйте задать небольшую скорость по Z:
M203 Z5 (тут значения в мм/с)
А потом отправить его по Z с большой скоростью:
G0 Z100 F7800
Ограничения скорости срабатывают?
************************
ОТВЕЧАЮ:
Лимиты
M203 X120.00
M203 Y120.00
M203 Z10.00
M203 E40.00
M201 X1600.00
M201 Y1600.00
M201 Z20.00
****************
Далее - попробовал дать команды:
M203 Z5
и
G0 Z100 F7800
Ограничения скорости срабатывают нормально. Это прикольно. Потому как при попытке напечатать любую модель, даже без М900 в коде, ось Z вырубается после прихода стола к точке начала печати. Вся печать идет в первый слой и движок не на удержании - просто пустой. Победить настройками ограничения скоростей невозможно.
Счас печатаю на 2.09. 
Модели без ЛА печатаются нормально. Но если в коде есть М900, то ось Z также вырубается после прихода стола к точке начала печати.

PlasmaKirill, 2024-02-08:
Ребят подскажите - где взять нормальную прошивку или "на крайняк" стоковую на Гостя5 с материнкой Reborn v.3.0 с распаянными tmc2225?

Sergey1560, 2024-02-08:
Стандартные прошивки есть тут 
https://github.com/Sergey1560/fb4s_howto/blob/master/mks_board/readme.md

Как изменить файл конфигурации под ваши драйвера описано тут:
https://github.com/Sergey1560/fb4s_howto/blob/master/stock_fw/readme.md

PlasmaKirill, 2024-02-09:
Здравствуйте, Sergey1560,

в разделе платы Flying Bear Reborn v3.0 в строке"
* Стандартная прошивка: отсутствует ссылка на прошивку вообще!
А в ковырянии недр самой прошивки я не силен.  (((
Спасибо что ответили.

-- 
Best regards,
Кирилл Иванов